MrChromebox/scripts #107: chromebook pixel 2013,there is no sound output when plug in earphone

canserliao, 2020-01-21:
In chromebook pixel 2013,there is no sound output when plug in earphone. Sound is still output from speaker in cloudready image. Even I use ubuntu to check if this is an OS issue. But it is still fail in ubuntu.

MrChromebox, 2020-01-21:
this isn't an issue with my script, nor my firmware, best I can tell. 

It works under ChromeOS, so has to be something different there vs Cloudready or mainline Linux.

closed 2020-01-21 by MrChromebox

canserliao, 2020-02-14:
Thanks for the reply!
You meant your firmware can work under chromeOS; so after run the script how to install chromeOS (or which chromeOS can be used to install?)

MrChromebox, 2020-02-14:
one cannot run ChromeOS with the UEFI firmware. you need to use CloudReady, or a hack solution like Chromefy / project croissant