MrChromebox/scripts #109: Unable to enter legacybios

mylovesaber, 2020-03-01:
>Model：hp chromebook x360 14
>Original chromeos and bios

I installed the legacy bios and it showed installation is successful.Then I set the following command in root mode:
```
crossystem dev_boot_legacy=1
enable_dev_usb_boot
```
When I rebooted and press 'CTRL+L',nothing happened except hearing two beeps and the TAB shows below:
```
HWID: SONA A2B-A2C-C3F-N6C-I9F
recovery_reason: 0x00 / 0x00 / 0x00 Recovery not requested
VbSD.flags: 0x0003dc54
VbNv.raw: 60 10 00 00 05 02 00 20 00 fe ff 00 00 ff ff 7d
dev_boot_usb: 1
dev_boot_legacy: 1
dev_default_boot: 0
dev_boot_signed_only: 0
dev_boot_fastboot_fll_cap: 0
TPM: fwver=0x00010001 kernver=0x00010001
gbb.flags: 0x00000000
gbb.rootkey:787b5d1f1a86d5d3d536f45d682ebb3374641747
gbb.recovery_key: 7285e9db60f3b90ff3e14ba2c1782664b1f995
kernel_subkey: 2dd17999d27cb66d1f1af9b748201e5f6af8f645
read-only firmware id: Google_Nami.10775.123.0
active firmware id: Google_Nami.10775.123.0
TPM state: v=1 failed tries=0 max_tries=200
```
https://mrchromebox.tech/#chromeos

mylovesaber, 2020-03-01:
The same as ：
https://www.reddit.com/r/chromeos/comments/fbnzmj/hp_chromebook_x360_14_cant_enter_legacybios/

MrChromebox, 2020-03-02:
should be fixed now

closed 2020-03-02 by MrChromebox