MrChromebox/scripts #110: Chromebook Pixel 2013 (LINK)

badrock21, 2020-03-24:
I ran your script today and it couldn't download the firmware to flash back to factory.  Luckily I had a usb backup from a year or 2 ago.  Is this something hosted here or via google as I know that google no longer has the image for the link chromebook on their dl recovery script.

MrChromebox, 2020-03-24:
accidentally broke it earlier today, fixed now

MrChromebox, 2020-03-24:
curious, why would you go back to stock firmware on LINK given it's EOL ?

closed 2020-03-24 by MrChromebox

badrock21, 2020-03-24:
With all the homeschool stuff going on right now my wife finds it easier to use stock as the zoom audio works better.  If I use cloudready I usually have to go in to alsa mixer to get sound to work properly so for now I just flashed it back to stock.  Thanks and thankyou for all your work.  I still like the 2013 chromebook pixel has as good a screen as anything out there.