MrChromebox/scripts #112: Password required on crosh>shell ?

disizvince, 2020-04-25:
Hello (sorry for my english)
I am trying to download firmwire Utility Script but it asks me a password, why ? Which one ?
Thx for your help :

crosh> shell
chronos@localhost / $ cd; curl -LO https://mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   185    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  1078  100  1078    0     0   1398      0 --:--:-- --:--:-- --:--:--  1398
We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:
    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.
Password: 

[image]

MrChromebox, 2020-04-25:
there isn't a password unless you incorrectly enabled debugging options
when switching to developer mode. exit/re-enable developer mode and there
will be no password

disizvince, 2020-04-26:
You're right, thanks.
Another Issue, I do not find on your website how to disable "write-protect" which allow to install your FullROM.
Any clue, please?

MrChromebox, 2020-04-26:
on https://mrchromebox.tech/#devices there is an entry for each device
which lists (and sometimes links to) the write protect method used

disizvince, 2020-04-27:
OK, thanks a lot.
It is a screw for ASUS C200M.
I did not find a picture about which one, so I try this screw and it is working : 
[image]

AngelEyes311, 2020-05-04:
i thought i wanted to get mint...went through your whole tutorial ....with the password bc i enabled debugging and i didn't brick my system! but my usb boot didn't work, will be installing your recommendation GalliumOS, I totally thought you were speaking to me w the warnings :) thank you for dumbin it down just enough! (and if i might add to the above post...its actually a physical screw?!?! lol)

closed 2020-05-04 by MrChromebox