MrChromebox/scripts #117: Stock firmware backup confirmed, but nowhere to be found (written to EFI partition!)

jjabba, 2020-05-17:
I backed the stock firmware of a HP Chromebox yesterday. The usb stick I had around was a FAT32 formatted one, but alas as it turns out, initially formatted using OS X...

Backup if the firmware reported success, so after removing the stick I proceeded to patch the full firmware which went as expected.

But when plugging the USB stick into my main Computer (Mac) the firmware file was nowhere to be found. Luckily I've been around the block a few times so I proceeded to investigate what exactly `firmware.sh`does as well as looking at the USB stick using testdisk. As it turns out, the firmware had been written, not to the mountable part of the stick, but to the hidden first EFI partition(!). 

In the end, testdisk was able to successfully copy the file off the EFI partition but I doubt the average user would know to try such methods if the firmware file isn't there despite successful backup and confirmation from the script.

[image]

MrChromebox, 2020-05-18:
the script backs up to the first mountable partition available on the device, in this case the FAT32 EFS partition. I don't have any logic to check the partition type, status, etc. Just checks for a mountable filesystem.

jjabba, 2020-05-19:
My point is that the user experience isn't what one might expect. And I fear there is a risk of stock firmware loss should the USB drive used have been formatted as FAT32 on a MacOS machine. Would you accept a patch which favors the largest partition rather than the first?

MrChromebox, 2020-05-19:
I'd consider it, but sounds like the real issue was MacOS not repartitioning the stick properly when wiping/formatting

jjabba, 2020-05-19:
I would agree, but I feel I might have a better chance for you accepting a PR than apple 😅

MrChromebox, 2020-05-19:
how about I simply skip hidden partitions and/or the EFS? Then if the next partition fails to mount due to incompatible FS, the user will be alerted

jjabba, 2020-05-21:
That sound's like an awesome solution to me!

MrChromebox, 2020-05-21:
I don't have time to implement this ATM, but would certainly take a pull request to implement it :)