MrChromebox/scripts #123: SD Card Detection when restoring stock and backing up firmware.

Shaddow798, 2020-07-10:
I found a bug when i tried to use a SD card with one partition to restore my stock firmware.

The issue is the inbuilt SD card reader default drive letter is /dev/mmcblk0 (the internal EMMC is /dev/mmcblk1) causes it not to be detected as the firmware utility script only scans /dev/sd* for devices.

MrChromebox, 2020-07-10:
problem is script assumes mmcblk0 is internal, there's no good way to differentiate between internal/external devices AFAIK.

Shaddow798, 2020-07-10:
@MrChromebox that's what I thought was just asking if there was another way that I couldn't think of. SMART data?
Probably to big of a pain to add for the one person who decides to us a SD for onse.