MrChromebox/scripts #125: Comet Lake Processor Support

leifliddy, 2020-07-12:
I'm considering buying Samsung Galaxy Chromebook to run Linux on (not a fan of Chrome OS and I don't care about long battery life), although it doesn't look like the comet lake chipset is supported yet. 

I'd just like to get your thoughts on it. Do you feel comet lake will be supported in the future?
I'm not sure how this firmware/bios is developed. Do you develop it yourself? Do you need a device to test on?
If that's the case I can probably work something out to get you a Samsung Galaxy Chromebook to test on. 

There's only a few 12-13 inch fanless laptops out there with a 2k (or higher) screen resolution. 
The Galaxy Chromebook laptop just checks all the boxes for me, it's a shame it's not supported yet. 
Just let me know if there's anything I can do to get this device supported. 

Also, I'm a big fan of your project, and I truly appreciate all of the work you've done so thus far. Thanks!

MrChromebox, 2020-07-12:
> I'm considering getting a Samsung Galaxy Chromebook to run Linux on

that's a bad idea unless you're a developer, and/or don't mind things not working. No 2017+ Chromebooks work OOTB with native Linux, and most have significant functional deficiencies (audio, tablet mode, pen input, etc). Some of it is upstream drivers, some of it is ALSA UCM, config etc.

> Do you feel comet lake will be supported in the future?

Legacy Boot Mode is broken on Google's end, working with them to fix. UEFI firmware is under development/testing.

> I'm not sure how this firmware/bios is developed. Do you develop it yourself? Do you need a device to test on?

yes. yes.

> If that's the case I can probably work something out to get you a Samsung Galaxy Chromebook to test on.

drop me an email at mrchromebox at gmail

> Also, I'm a big fan of your project, and I truly appreciate all of the work you've done so thus far. Thanks!

TYVM

leifliddy, 2020-07-12:
> most have significant functional deficiencies (audio, tablet mode, pen input, etc). Some of it is upstream drivers, some of it is ALSA UCM, config etc.
I don't really care about tablet mode, pen input, or touchscreen functionality. 

I have some experience with the audio subsystem
I was able to work with some kernel developers, and over the course of two years managed to the audio working on the macbook 12-inch models.
https://github.com/leifliddy/macbook12-audio-driver

I'm just looking for base functionality, everything else can be sorted out later. 

> Legacy Boot Mode is broken on Google's end, working with them to fix. UEFI firmware is under development/testing.
  
Don't have any need for Legacy mode, just interested in UEFI-based firmware. 

I'll seriously consider procuring this laptop sometime next month (on vacation ATM), I'll keep you updated.

MrChromebox, 2021-10-01:
UEFI Full ROM firmware for CML was added for the 7/25/2021 release

closed 2021-10-01 by MrChromebox