MrChromebox/scripts #126: USB ports no longer powered during boot after upgrade from 4.10 to 4.12

simulationrose, 2020-07-18:
After upgrading the script from 4.10 to 4.12 on my ASUS Chromebox 2, I'm no longer able to access the boot menu via "ESC" (the white rabbit splash screen is visible) because the USB keyboard gets no power during the boot process anymore. I see this behavior on all 5 USB ports of the box and was able to confirm the behavior with a bootable USB memory stick which is no longer detected and starts flashing only when I already see the Linux login screen. While I was using version 4.10 the keyboard and the USB stick were available right after switching on the box. Now, the USB devices start working after (!) the boot sequence is over. Please have the kindness to have a quick look at this issue. Thx a lot.

MrChromebox, 2020-07-18:
I'm not seeing this on my test CN62 box here. Keyboard works fine, menus work fine, boot menu is populated with any attached USB devices

simulationrose, 2020-07-18:
Thx for the quick answer. I did some more experimenting ...
I think the issue is that I used a USB-C monitor which is also powered by the same USB-C cable. As soon as I change from USB-C monitor to an HDMI monitor the USB drive flashes right from the start ...
But now I have a different issue: on my HDMI 4K monitor I only see only thin white lines instead of the normal splash screen. I guess, the boot manager has some problems with this high 4K resolution. :-(
Unfortunately, I have only this 4K monitor (USB ports working, but no splash screen & menu) or the USB-C powered HD monitor (where the USB-A ports don't work until the boot sequence is over). 
In conclusion: I will get another HDMI HD monitor to do the setup work and everything should be fine ...

MrChromebox, 2020-07-18:
USB-C? so we're talking about the CN65 / Chromebox 3 / TEEMO then

simulationrose, 2020-07-18:
Yes, you're right ... My fault ... Sorry ...

simulationrose, 2020-07-18:
Solved the issue by using my TV set as monitor during the setup ... Everything is fine now ... Issue can be closed ... Thx for your help.

closed 2020-07-18 by MrChromebox