MrChromebox/scripts #128: Can't install UEFI Firmware to Google Pixelbook using firmware-util.sh

soultoru, 2020-07-26:
Hi! I tried to install UEFI using firmware-util.sh, but failed with flashrom. How can I solve this problem?

---
#### The model and OS

* Google Pixelbook 2017 Core i7, 16 GB RAM, 512 GB
* Google Chrome OS: Version 84.0.4147.94 (Official Build) (64-bit)

#### Displayed firmware information 

```
 ChromeOS Device Firmware Utility Script [2020-07-18] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**   Device: Google Pixelbook (EVE)
** Platform: Intel KabyLake
**  Fw Type: Stock ChromeOS w/RW_LEGACY
**   Fw Ver: Google_Eve.9584.201.0 (11/18/2019)
**    Fw WP: Disabled
*********************************************************
```

#### Flashrom's log
[flashrom.log](https://github.com/MrChromebox/scripts/files/4977735/flashrom.log)

#### Others
I referred to [this repository](https://github.com/yusefnapora/pixelbook-linux) for instructions on how to run firmware-util.sh on Pixelbook.

MrChromebox, 2020-07-26:
not a script issue, it's an issue of using outdated instructions instead of mine

see note at bottom of: https://wiki.mrchromebox.tech/Firmware_Write_Protect#Disable_WP_.2F_Enable_Firmware_Flashing

closed 2020-07-26 by MrChromebox

soultoru, 2020-07-27:
Thank you for your reply. I was successful with reboot and restore.

MrChromebox, 2020-07-27:
I'll try to add a check to the script for CR50 devices to ensure they've been rebooted, to avoid this issue and the hassle of performing a recovery