MrChromebox/scripts #130: dual boot on Edgar is erratic

mathnic, 2020-08-24:
I happily single booted gallium on my Acer/Edgar machine for many years (THANKS for the firmware!!!), then a month ago I added an external USB/SSD to try out Kubuntu. At that time I also updated the firmware to the latest, June 6th version.
Kubuntu turned out to work really nicely out of the box and it became my daily driver on the USB drive. However, I've experienced some very odd behavior on boot up and I can't see any consistent pattern... More a nuisance in the beginning but now feels like I've had enough and will probably remove gallium and hopefully return to reliable/functional single boot. Before I do, I thought I ought to check in here to see if I can be helpful tracking down the problem. 
It's hard for me to pinpoint what's going on, but here are some observations:
1. There seems to be a cross dependence between kubuntu(SSD) and gallium(mmc). Depending on how I boot and shut down gallium seems to affect the boot up process/options for a subsequent kubuntu boot.
2. Sometimes the boot process seems to hang and after power reboot, the previous boot order is gone. 
3. After resetting the boot order to USB first, I can't directly boot that. If I then start up gallium and restart, voila, now the boot order is right and kubuntu starts!
4. Most of the time (i.e. erratic) in kubuntu, media streaming will get hung with buffering problems. I'd shut down, start gallium, shut down, restart kubuntu after the usual inconsistent booting and then media streaming works great! Most of the time, not so much. I see no pattern. How in the world does this relate to BIOS??? It just seems like it does. Like some weird memory leak? 
5. Oddly (at least to me), I've noticed that the efi FAT partition on the mmc drive is mounted when I run kubuntu. I would have expected the efi partion on the SSD to be used. However, it seems it is never mounted, regardless if I'm running kubuntu or gallium. Perhaps this is completely normal...

Anyway, as I said, I'm ready to dump gallium and single boot kubuntu, but if there is anything I can do to help the cause, I'll do it.

MrChromebox, 2020-08-24:
honestly, I have no time or inclination to support legacy boot stuff in 2020. If you're not running ChromeOS, the UEFI firmware is a better, more flexible option -- especially for what you want to do

MrChromebox, 2020-08-24:
also, as this isn't a script issue, would request that you move it to firmware instead :)

closed 2020-08-24 by MrChromebox