MrChromebox/scripts #131: Download firmware-util.sh error

TM-001, 2020-09-04:
run  `curl -LO mrchromebox.tech/firmware-util.sh` 
download file error 

(checked carefully  command is right)

MrChromebox, 2020-09-04:
things fail to download all of the time, could be DNS, could be github, could be any number of things. Without the error there's no way to say, and even then.

closed 2020-09-04 by MrChromebox

cshirey1980, 2020-09-04:
Okay well here is my latest and greatest in this issue I was wanting to check out cloud ready so I downloaded it on the USB with their writer after attempting to load it and restart black screen of death has arrived . You can see the screen is lit but there is nothing on it no cursor no prompt no nothing and every key that I hit every anything that I try makes no difference

cshirey1980, 2020-09-05:
Well like I said the screenshots completely black so there's nothing to really show period it was the 2013 pixel so it might just be the end of the road for it but it was working fine before I tried to do cloud ready I don't know what happened

TM-001, 2020-09-05:
Yes, the problem is internet. I can't download directly file by github. You know, some locate internet is limited.
Thanks for reply.