MrChromebox/scripts #133: Recover MAC without firmware/VPD backup?

kirk444, 2020-10-16:
A (very) long time ago, I used the tool to replace the firmware on an Asus CN60 Panther. At the time, no copy was kept of the original firmware, specifically the VPD region. Fast forward to today, I recently flashed the latest full-UEFI firmware and realized that I have a generic MAC (a0:00:ba:d0:0b:ad) as a result. Is there any way to recover the original MAC address for my device, aside from that VPD region?

MrChromebox, 2020-10-16:
recover? no, it's gone. If you have the MAC from somewhere else (bottom sticker, box, etc) or want to make up one, you can easily inject one into the firmware as part of an update, but there's no way to recover the original VPD region if you didn't back up and flashed a (very) old version of the firmware which didn't preserve it (or someone else's firmware(

kirk444, 2020-10-16:
Thanks, I've torn the device (Asus Chromebox CN60) down and it doesn't seem to have any stickers/labels with the MAC (internal or external). I think I originally flashed with John Lewis' firmware, but it was many years ago.

Thanks for your repsonse, and for everything else you've done for the community.

closed 2020-10-16 by kirk444

cab404, 2023-05-15:
Hi @MrChromebox!

And if I want to set it in hardware, what should I do?

I SWD-d firmware onto the thing, and saved VPD old firmware, yet writing that region doesn't fix MAC — showing AD:00:BA:D0:0B:AD :(

(We have multiple machines on one network, and that really messes with us)

[image]
_lesspipe of vpc region_

Google VPD doc tells something about "ethernet_mac" key, and I didn't test setting it yet (probably the next thing to try out — maybe you know some tried and true method not requiring further messing with SWD programmer?

MrChromebox, 2023-05-15:
> And if I want to set it in hardware, what should I do?

you can't, to my knowledge - only set it in firmware.

>  yet writing that region doesn't fix MAC

if the region doesn't contain the ethernet_mac key and value, then that's why. You can insert it using google's VPD tool, if you still have it from the product box, or just make them up. The VPD region you've shown above is missing more than just the ethernet_mac key though.

If you want to drop me an email with the saved VPD file, or original device firmware, I can take a look at it