MrChromebox/scripts #134: It says I need a password when I run cd ~; curl -L -O http://mrchromebox.tech/firmware-util.sh; sudo bash firmware-util.sh

bruhmoment1234567, 2020-10-28:
When I run cd ~; curl -L -O http://mrchromebox.tech/firmware-util.sh; sudo bash firmware-util.sh, it says I need a password from the local system administrator. Here is an image of the problem.

[image]

MrChromebox, 2020-10-28:
there's only a password if you set one, or enabled debugging features when you shouldn't have. there is no sudo password by default

bruhmoment1234567, 2020-10-29:
The thing is, I never set a password.

bruhmoment1234567, 2020-10-29:
And I don't have debugging features enabled.

MrChromebox, 2020-10-29:
dunno what to tell you, there's not one by default. Exit and re-enter developer mode and don't click/enable anything not explicitly stated to do so

bruhmoment1234567, 2020-10-29:
Ok.

closed 2021-10-01 by MrChromebox