MrChromebox/scripts #136: Unable to read current firmware, cannot reflash factory ROM backup

duck955, 2020-11-05:
I flashed my Lenovo Chromebook AMD A6 to the UEFI BIOS and was able to install different OSs and boot from multiple devices.  The chromebook is the model released in September with the Treeya motherboard and Stoney Ridge processor

I want to return to ChromeOS or crx and used the Linux script from the website.  When I run it, i receive the error that it cannot read the firmware and it asks me to use iomem=relaxed or use a different kernel.  I tried both items at the same and different times.  I've run Kali and Ubuntu 19.1/20.04, on Live USB and installed.

The script was able to overwrite my current BIOS, it is having issues to returning it to stock.  I have the stock .ROM image backed up.  

What can I do to return to stock/factory BIOS so I can install ChromeOS?

[image]
[image]
[image]
[image]

MrChromebox, 2020-11-05:
add ` iomem=relaxed` after `quiet splash` (but inside the end quotes), then run `sudo update-grub` and reboot. After that, the script should run without issue

duck955, 2020-11-05:
I appreciate the quick reply.  I posted the pic of the wrong grub, I have used the iomem=relaxed and it still does not work, script fails with the same error.

[kern.log](https://github.com/MrChromebox/scripts/files/5491966/kern.log)

[image]

duck955, 2020-11-05:
To convolute the issue more, if I could get the AMD chipset drivers working correctly I would have no reason to go back.  Right now, I have no Function Key control, brightness control, or sound with any of the other distros, Kali, Ubuntu, or XBuntu.

duck955, 2020-11-18:
I continue to have this issue, even with a fresh install of Ubuntu 20.04.1 LTS with the iomem=relaxed line in the grub loader.  What other logs can I provide?

MrChromebox, 2020-11-18:
run my script, and after it fails, run `sudo /tmp/flashrom -p host -V`  and report the output (pastebin works well)

duck955, 2020-11-18:
Thank you for the additional help, I hope log is in the form you requested.  I downloaded it from 

<script src="https://pastebin.com/embed_js/iVga87DW"></script>

MrChromebox, 2020-11-18:
> Thank you for the additional help, I hope log is in the form you requested.
> 
> [Pastebin](https://pastebin.com/embed_js/iVga87DW)

thanks. here's the problem:

> Found Generic flash chip "unknown SPI chip (RDID)" (0 kB, SPI) at physical address 0x0

it thinks the flash chip is size 0, so nothing to read. I'm guessing the version of flashrom I'm using with my script is too old.  I've just updated it to a newer build, so try running the script again. If it fails, please run the command above again and pastebin the output.

duck955, 2020-11-18:
Thank for reviewing the logs, I ran the script again and the output is linked.

[Pastebin-NewScript](https://pastebin.com/embed_js/XWSSF4GK)

MrChromebox, 2020-11-18:
the script did not download the new flashrom build; `sudo rm /tmp/flashrom` and run script again

duck955, 2020-11-18:
It worked!!  I deleted the firmware-util script from the home directory, I was unaware of the file in the tmp folder.

Thank you for your help, hopefully this will help others with newer chromebooks too.

MrChromebox, 2020-11-18:
great, glad that resolved the issue

closed 2020-11-18 by MrChromebox

tlewiscpp, 2022-03-23:
> I flashed my Lenovo Chromebook AMD A6 to the UEFI BIOS and was able to install different OSs and boot from multiple devices. The chromebook is the model released in September with the Treeya [...]

Hey, sorry to bring this back up. @duck955 , I have also flashed the custom UEzfI firmware, and I want to return to stock. I put the backup of the firmware but accidentally overwrote it. Would you be able to share the stock firmware (or back it up again)? Thanks for any help you can provide.

lunaboy1973, 2023-02-02:
I have the exact same problem: Unable to read current firmware; cannot continue.

When trying to run the script.

I am using a Acer Chromebook CB514-1HT-P0BX with Linux Mint 21.1. I was able to flash the BIOS with your script after disconnecting the battery and thereby disabling write protection. 
But whenever I now run it, I receive this error message. Adding iomega=relaxed in grub did not help.
sudo /tmp/flashrom -p host -V returns this:
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 5.19.0-28-generic (x86_64)
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 5.19.0-28-generic (x86_64)
flashrom was built with unknown PCI library, GCC 7.5.0, big endian
Command line (3 args): /tmp/flashrom -p host -V
Acquiring lock (timeout=180 sec)...
Opened file lock "/run/lock/firmware_utility_lock"
Lock acquired.
Initializing internal programmer
Found candidate at: 00000500-00000528
Found coreboot table at 0x00000500.
Found candidate at: 00000000-00000534
Found coreboot table at 0x00000000.
coreboot table found at 0x7ab22000.
coreboot header(24) checksum: 383d table(1308) checksum: 30ce entries: 43
Vendor ID: Google, part ID: Coral
Using External DMI decoder.
DMI string chassis-type: "Laptop"
Laptop detected via DMI.
DMI string system-manufacturer: "Google"
DMI string system-product-name: "Coral"
DMI string system-version: "rev3"
DMI string baseboard-manufacturer: "Google"
DMI string baseboard-product-name: "Coral"
DMI string baseboard-version: "rev3"
get_target_bus_from_chipset() returns 0x10.
Found chipset "Intel Apollolake" with PCI ID 8086:5af0. Enabling flash write... Vendor ID: 8086, Device ID: 5a96, BAR: 91b15000

BIOS_CNTL = 0x8b: BIOS Lock Enable: enabled, BIOS Write Enable: enabled
SPI BAR is = 0x91b15000
Error accessing PCH SPI BAR0, 0x4000 bytes at 0x0000000091b15000
/dev/mem mmap failed: Resource temporarily unavailable
FAILED!
FATAL ERROR!
Error: Programmer initialization failed.
restore_power_management: Re-enabling power management.

Please help. And thank you so much for your great work!

MrChromebox, 2023-02-02:
> Please help. Do I need to disconnect battery whenever I want to run your script?

absolutely not, the kernel is just preventing flashrom from reading/writing the firmware. I haven't tested Mint 21.1 yet, but  I'm guessing it needs a newer build of flashrom. Try live booting a 2204-based distro and updating from there

lunaboy1973, 2023-02-02:
Trying to update from 21.1 Mint live distro executes the script, but it fails to recognize platform. Device is reported as "Error: unknown or unidentifiable device (CORAL)"

It's weird because I managed to successfully flash the BIOS with your script in the first place. First the legacy, then the UEFI.

MrChromebox, 2023-02-02:
what's the output of `sudo dmidecode -s system-product-name` ?
what is the correct board name? EPAULETTE?

lunaboy1973, 2023-02-02:
sudo dmidecode -s system-product-name returns "Coral"

Yes, it is Epaulette

MrChromebox, 2023-02-02:
I'm not sure why your device either got flashed with the wrong firmware image, or isn't reporting the device name correctly. You can pastebin a cbmem log for me?
https://mrchromebox.tech/files/util/cbmem.tar.gz - extract, then run:
`sudo ./cbmem -1 > cbmem.log` and pastebin the log and provide URL here

lunaboy1973, 2023-02-02:
[NOTE ]  coreboot-4.19-120-gfc756557e50-MrChromebox-4.19 Mon Jan 23 19:54:11 UTC 2023 bootblock starting (log level: 7)...
[DEBUG]  LPC: Opened IO window LGIR0: base 800 size 100
[DEBUG]  LPC: Opened IO window LGIR1: base 900 size 100
[DEBUG]  FMAP: Found "FLASH" version 1.1 at 0xab5000.
[DEBUG]  FMAP: base = 0x0 size = 0x1000000 #areas = 14
[DEBUG]  FMAP: area COREBOOT found @ ab5300 (4697344 bytes)
[DEBUG]  FMAP: area COREBOOT found @ ab5300 (4697344 bytes)
[INFO ]  CBFS: mcache @0xfef04e00 built for 26 files, used 0x590 of 0x4000 bytes
[INFO ]  CBFS: Found 'fallback/romstage' @0x80 size 0x8a19 in mcache @0xfef04e2c
[DEBUG]  BS: bootblock times (exec / console): total (unknown) / 0 ms

[NOTE ]  coreboot-4.19-120-gfc756557e50-MrChromebox-4.19 Mon Jan 23 19:54:11 UTC 2023 romstage starting (log level: 7)...
[INFO ]  CPU: Intel(R) Pentium(R) CPU N4200 @ 1.10GHz
[INFO ]  CPU: ID 506c9, Apollolake B0, ucode: 0000002c
[INFO ]  CPU: AES Supported, TXT Not Supported, VT Supported
[INFO ]  MCH: device id 5af0 (rev 0b) is Apollolake
[INFO ]  PCH: device id 5ae8 (rev 0b) is Apollolake
[INFO ]  IGD: device id 5a84 (rev 0b) is Apollolake HD 505
[WARN ]  HECI: CSE device 0f.1 is disabled
[WARN ]  HECI: CSE device 0f.2 is disabled
[DEBUG]  pm1_sts: 0000 pm1_en: 0000 pm1_cnt: 00001800
[DEBUG]  gpe0_sts[0]: 00000000 gpe0_en[0]: 00000000
[DEBUG]  gpe0_sts[1]: 10000000 gpe0_en[1]: 00000000
[DEBUG]  gpe0_sts[2]: 00000200 gpe0_en[2]: 00000000
[DEBUG]  gpe0_sts[3]: 00000000 gpe0_en[3]: 00000000
[DEBUG]  prsts: 00000000
[DEBUG]  tco_sts:   0000 0000
[DEBUG]  gen_pmcon1: 08004000 gen_pmcon2: 00003a00 gen_pmcon3: 00000000
[DEBUG]  prev_sleep_state 5
[DEBUG]  FMAP: area COREBOOT found @ ab5300 (4697344 bytes)
[DEBUG]  FMAP: area COREBOOT found @ ab5300 (4697344 bytes)
[INFO ]  CBFS: Found 'fspm.bin' @0x51380 size 0x296c3 in mcache @0xfef050b8
[DEBUG]  FMAP: area RW_MRC_CACHE found @ aa0000 (65536 bytes)
[INFO ]  LPDDR4 SKU id = 0x4
[INFO ]  LP4DDR speed is 2400MHz
[INFO ]  LPDDR4 Ch0 density = 8 Gb
[INFO ]  LPDDR4 Ch1 density = 8 Gb
[DEBUG]  FMAP: area RW_VAR_MRC_CACHE found @ ab0000 (4096 bytes)
[DEBUG]  CBMEM:
[DEBUG]  IMD: root @ 0x7afff000 254 entries.
[DEBUG]  IMD: root @ 0x7affec00 62 entries.
[DEBUG]  FMAP: area RO_VPD found @ ab1000 (16384 bytes)
[ERROR]  init_vpd_rdev: No RW_VPD FMAP section.
[DEBUG]  External stage cache:
[DEBUG]  IMD: root @ 0x7b7ff000 254 entries.
[DEBUG]  IMD: root @ 0x7b7fec00 62 entries.
[DEBUG]  CPU: frequency set to 2500 MHz
[DEBUG]  FMAP: area RW_VAR_MRC_CACHE found @ ab0000 (4096 bytes)
[DEBUG]  MRC: Checking cached data update for 'RW_VAR_MRC_CACHE'.
[INFO ]  SF: Detected 00 0000 with sector size 0x1000, total 0x1000000
[DEBUG]  MRC: cache data 'RW_VAR_MRC_CACHE' needs update.
[DEBUG]  MRC: updated 'RW_VAR_MRC_CACHE'.
[DEBUG]  4 DIMMs found
[DEBUG]  SMM Memory Map
[DEBUG]  SMRAM       : 0x7b000000 0x800000
[DEBUG]   Subregion 0: 0x7b000000 0x700000
[DEBUG]   Subregion 1: 0x7b700000 0x100000
[DEBUG]   Subregion 2: 0x7b800000 0x0
[DEBUG]  top_of_ram = 0x7b000000
[DEBUG]  Normal boot
[INFO ]  CBFS: Found 'fallback/postcar' @0xa6040 size 0x5d84 in mcache @0xfef05140
[DEBUG]  Loading module at 0x7abcf000 with entry 0x7abcf031. filesize: 0x59a0 memsize: 0xbd10
[DEBUG]  Processing 233 relocs. Offset value of 0x78bcf000
[DEBUG]  BS: romstage times (exec / console): total (unknown) / 1 ms

[NOTE ]  coreboot-4.19-120-gfc756557e50-MrChromebox-4.19 Mon Jan 23 19:54:11 UTC 2023 postcar starting (log level: 7)...
[DEBUG]  Normal boot
[DEBUG]  FMAP: area COREBOOT found @ ab5300 (4697344 bytes)
[DEBUG]  FMAP: area COREBOOT found @ ab5300 (4697344 bytes)
[INFO ]  CBFS: Found 'fallback/ramstage' @0xcb80 size 0x1da91 in mcache @0x7abdd0dc
[... 2279 lines cut ...]
[DEBUG]    Entry Point 0x00801626
[DEBUG]  BS: BS_PAYLOAD_LOAD run times (exec / console): 228 / 0 ms
[DEBUG]  CSE FWSTS1: 0x80000245
[DEBUG]  CSE FWSTS2: 0x89100000
[DEBUG]  CSE FWSTS3: 0x00000000
[DEBUG]  CSE FWSTS4: 0x00080000
[DEBUG]  CSE FWSTS5: 0x00000000
[DEBUG]  CSE FWSTS6: 0x40000000
[DEBUG]  CSE: Working State          : 5
[DEBUG]  CSE: Manufacturing Mode     : NO
[DEBUG]  CSE: Operation State        : 1
[DEBUG]  CSE: FW Init Complete       : YES
[DEBUG]  CSE: Error Code             : 0
[DEBUG]  CSE: Operation Mode         : 0
[DEBUG]  CSE: IBB Verification Result: FAIL
[DEBUG]  CSE: IBB Verification Done  : NO
[DEBUG]  CSE: Actual IBB Size        : 0
[DEBUG]  CSE: Verified Boot Valid    : FAIL
[DEBUG]  CSE: Verified Boot Test     : NO
[DEBUG]  CSE: FPF status             : FUSED
[DEBUG]  CSE: Error Status Code      : 0
[INFO ]  Disabling Heci using PCR
[DEBUG]  BS: BS_PAYLOAD_LOAD exit times (exec / console): 15 / 0 ms
[DEBUG]  mp_park_aps done after 0 msecs.
[DEBUG]  Jumping to boot code at 0x00801626(0x7ab22000)

lunaboy1973, 2023-02-02:
sorry

[cbmem.log](https://github.com/MrChromebox/scripts/files/10572330/cbmem.log)

MrChromebox, 2023-02-02:
unfortunately there's nothing in there that tells me why your device is being misidentified. I can have the script accommodate a CORAL device (which no Chromebook should show up as), but that doesn't solve the problem really

lunaboy1973, 2023-02-03:
Hmm. Really weird. Because it was identified successfully the first time and your BIOS runs fine on my device. So I guess I won't be able to update then...

MrChromebox, 2023-02-03:
drop me an email / ping me on discord, I want to manually flash the correct file on your device and see if it's still misidentified

cmi021, 2023-03-27:
[image]

Need help with this unable to proceed

MrChromebox, 2023-03-27:
boot a live USB of a newer distro (22xx) and run the script from there

cmi021, 2023-03-27:
Thank you it works..

Bllckk, 2023-04-20:
What does newer distro (22xx) mean please? I have the same error but dont know how to do it or I dont understand your proffesional language

MrChromebox, 2023-04-20:
> What does newer distro (22xx) mean please?

it means use Ubuntu 2204 or 2210 (ie, a 2022 release) or similar

kaylteks, 2023-07-20:
[image]