MrChromebox/scripts #137: Custom tianocore build not retaining EFI variables

nariox, 2020-11-16:
Hello,
I am sorry if this is not the place to ask this. I have compiled a custom coreboot ROM for my C720 (cypress version) using the `build-uefi.sh` script. Everything, as far I can tell, is working properly. I can boot, suspend etc. The only thing not working is that EFI variables are reset everything I reboot the computer. `efivar-tester` does not report any issues, though.

I have obtained most blobs by extracting them from the most current ROM from the website and gotten the GOP driver from win-raid. I have had to downgrade my gcc to 8.3 to compile the toolchain, but once the toolchain is build, I assume my own gcc version shouldn't matter. The only change that I have applied was using a custom bootsplash image (90% of the reason for building the ROM by myself - my daughter wanted "a cuter bunny" in the boot sequence).

Am I missing some configuration? I have followed the steps that the `firmware.sh` script does, by disabling WP and setting the range to `0 0`. I am not saving the RW_MRC_CACHE or SMMSTORE since I did not find them in the original ROM.

Any help is greatly appreciated.

MrChromebox, 2020-11-16:
which coreboot branch are you building from? using my default config?

nariox, 2020-11-16:
I'm using the `2020.06.04` branch using the `.config.peppy.uefi` (although I've been simply calling `$ ./build-uefi.sh peppy`, which as far as I can tell does the same).

MrChromebox, 2020-11-16:
that's because I'm a dummy and forgot to set the default tianocore branch to 2020.06.04. the default branch (uefipayloadpkg) has since changed due to changes on the coreboot side to SMMSTORE v2, and is no longer compatible with the 2020.06.04 coreboot branch.

edit config.peppy.uefi and add this line to the end:

`CONFIG_TIANOCORE_REVISION_ID="origin/2020.06.04"`

then rebuild with `./build-uefi.sh peppy`

nariox, 2020-11-17:
You are amazing. Thank you so much!

nariox, 2020-11-18:
Now everything works. Closing!

closed 2020-11-18 by nariox