MrChromebox/scripts #138: UEFI sound not working.

ergpopler, 2020-11-25:
I was able to get the UEFI script thing to work, and boot into Kubuntu 20.04, and KDE Neon, both had no sound, via the internal speakers, or via the headphone jack, i tried fiddling with alsa and whatever, didn't work.  Bluetooth worked though, and was able to get sound to work with that, but it had horrible sound quality(i don't exactly know why).

MrChromebox, 2020-11-25:
this 1) isn't a script issue, and 2) isn't a firmware issue. It's a kernel/driver/Alsa UCM config one.

closed 2020-11-25 by MrChromebox

ergpopler, 2020-11-25:
Is it solvable?

MrChromebox, 2020-11-25:
depends on the device/platform, distro, kernel, etc.

GalliumOS has working audio for some older platforms. Newer devices don't work regardless.  Check out https://www.reddit.com/r/GalliumOS and https://www.reddit.com/r/chrultrabook