MrChromebox/scripts #143: /tmp/ can be mounted in noexec mode

patatetom, 2021-01-13:
hi,
this is the case on my CB Asus C301SA ;-)
thanks for your work !

```diff
--- functions.sh.org	2021-01-13 08:40:22.949997000 +0100
+++ functions.sh	2021-01-13 08:47:28.909996042 +0100
@@ -399,22 +399,34 @@
 #start with a known good state
 cleanup
 
-#get required tools
+#get required tools and check executability
 get_flashrom
 if [ $? -ne 0 ]; then
     echo_red "Unable to download flashrom utility; cannot continue"
     return 1
 fi
+if [ ! -x ${flashromcmd} ]; then
+    echo_red "Unable to execute flashrom utility; cannot continue"
+    return 1
+fi
 get_cbfstool
 if [ $? -ne 0 ]; then
     echo_red "Unable to download cbfstool utility; cannot continue"
     return 1
 fi
+if [ ! -x ${cbfstoolcmd} ]; then
+    echo_red "Unable to execute cbfstool utility; cannot continue"
+    return 1
+fi
 get_gbb_utility
 if [ $? -ne 0 ]; then
     echo_red "Unable to download gbb_utility utility; cannot continue"
     return 1
 fi
+if [ ! -x ${gbbutilitycmd} ]; then
+    echo_red "Unable to execute gbb_utility utility; cannot continue"
+    return 1
+fi
 
 #get full device info
 if [[ "$isChromeOS" = true && ! -d /sys/firmware/efi ]]; then
```

regards.

MrChromebox, 2021-01-13:
hmm, would it be better to simply check if /tmp mounted noexec and remount it?

patatetom, 2021-01-16:
hi,
why not ?
I haven't looked at the code in depth, but it seems that the commands can be taken elsewhere than in `/tmp/` in which case, testing the executability of each command may be preferable.
regards.

MrChromebox, 2021-01-16:
the latter two commands are always executed from the same location, so checking for both is redundant.
"unable to execute" is ambiguous, and lazy when we have the ability to both be more precise as to the issue, and to correct it without bothering the user at all

patatetom, 2021-01-18:
hi,
beyond their location, one of these two binaries (or even both) may have been disabled (`chmod -x`) for some reason : the three tests, even if they seem redundant, simultaneously address the different cases.
yes, the delivered message may be more explicit.
regards.