MrChromebox/scripts #145: Unable to read current firmware (Samsung Chromebook V2)

gomesf, 2021-01-23:
Hi, first of all, thanks for your amazing job and this script!

Unfortunately i'm having issues when trying to recover the Chrome OS.

Im using Ubuntu 16.04, kernel 4.4.0-194-generic, and the processor is and Intel Celeron 3965Y

I run the script and it shows the error:
```
Downloading supporting files...

Getting device/system info...

Unable to read current firmware; cannot continue.

Either add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu).

See https://www.flashrom.org/FAQ for more info

```
I already added the "iomem=relaxed" and "nopat" to the grub config (and i did run "sudo update-grub" after and restarted)

I tried to run, like i saw you suggesting in another topic, the command:
"sudo /tmp/flashrom -p host -V"

and the output is:
```
/tmp/flashrom: /lib/x86_64-linux-gnu/libpci.so.3: version `LIBPCI_3.5' not found (required by /tmp/flashrom)
/tmp/flashrom: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.27' not found (required by /tmp/flashrom)

```
Can you help me or give me light in this issue?

MrChromebox, 2021-01-23:
> Can you help me or give me light in this issue?

boot a modern Ubuntu version with a kernel that supports your hardware; I recommend 20.10. 16.04 is going to have issues, as you've seen first hand, due to not supporting your device's hardware

gomesf, 2021-01-25:
Hi! Using a newer version of ubuntu (20.04.01) worked! Thanks again for your support and patience!

closed 2021-01-25 by gomesf