MrChromebox/scripts #149: I know you said in the FAQ Windows 7 doesn't work...

higgy999, 2021-03-10:
...but what is wrong with it? I tried Windows 10 before on it and it ran painfully slow with the 2 core processor and all of the background tasks from Microsoft. I want to try it again with Windows 7, even though it's out of support, trackpad drivers don't work, etc. When booting with it from the Legacy Boot I get to the installer screen but it's either freezing or doesn't have drivers for keyboard, USB, or trackpad. I'm trying to fix it assuming the later. When doing a WinToGo installation with Rufus, it goes to the startup animation then I get a red screen of death extremely quickly without being able to read any errors. I want to know going farther with this if Windows 7 is even possible past the installation screen and what errors I'll encounter when doing it. The alternative is to use a custom version of Windows 10 with none of the tracking Microsoft does which I remember seeing somewhere on YouTube.

MrChromebox, 2021-03-10:
this is baffling. You ask what's wrong with it, then detail all the ways it doesn't work in the next few sentences. Windows 7 isn't supported because it doesn't work, and there's no one (with the required knowledge/skill) willing to put in the effort to make it work

closed 2021-03-10 by MrChromebox

JamieGrimwood, 2023-04-19:
It doesn't work because of UEFI (to the best of my knowledge) as chromebooks only support UEFI and windows 7 doesn't support it. It is quite an easy fix, just google online how to add UEFI to windows 7.

MrChromebox, 2023-04-19:
lack of UEFI support is not the reason Windows 7 isn't supported