MrChromebox/scripts #151: Only output to HDMI after changed a 1080p LCD

liyafe1997, 2021-04-01:
My chromebook cb3-532 came with 1366x768 internal display, and I changed to a 1080p one recently.

But I found all the firmware interfaces, include Ctrl+D developing mode (stock firmware), and the MrChromebox UEFI interface, only output to HDMI display, in those interfaces the internal one always stay at black. After entered an OS (whatever ChromeOS or Windows with graphics driver instaleld), the internal 1080p LCD can lights up.

So it is hard coded the screen timing(resolution) in the firmware, leads it unable to detect and output to the internal 1080p one? Any way to modify it to 1080p?

MrChromebox, 2021-04-01:
> So it is hard coded the screen timing(resolution) in the firmware, leads it unable to detect and output to the internal 1080p one? Any way to modify it to 1080p?

yes, the VBT (video BIOS tables) describe the supported panel resolutions, timings, settings, etc. are are specific to the board/panel. BANON never came with a 1080p panel, so the VBT likely doesn't contain compatible settings. One would need to know the panel model / settings / timings, then edit the VBT accordingly and re-flash the firmware

liyafe1997, 2021-04-01:
> yes, the VBT (video BIOS tables) describe the supported panel resolutions, timings, settings, etc. are are specific to the board/panel. BANON never came with a 1080p panel, so the VBT likely doesn't [...]

Thank you! Do you have any guide of how to do that?

MrChromebox, 2021-04-01:
> Thank you! Do you have any guide of how to do that?

nope. if you're lucky, you can find Intel's BMP tool to edit it, but also required is the mapping file (.bsf) specific to the platform and version of the VBT being edited.

MrChromebox, 2021-10-01:
closing as this isn't an issue with the script

closed 2021-10-01 by MrChromebox