MrChromebox/scripts #153: ChromeOS Firmware Utility Script Error

Chrome-GitHub, 2021-05-15:
Good Afternoon,

May I trouble for help with the following error message when running the Firmware Utility Script please?
cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                                        Dload Upload   Total   Spent      Left    Speed
  0        0    0               0  0        0            0          0    --:--:-- --:--:--     --:--:--          0
  0    185    0               0  0        0            0          0    --:--:-- --:--:--     --:--:--          0
100 1252 100       1252  0        0      2469          0    --:--:-- --:--:--     --:--:--     2469

Downloading supporting files...
firmware-util.sh: line 45: ./sources.sh: No such file or directory
firmware-util.sh: line 46: ./firmware.sh: No such file or directory
firmware-util.sh: line 47: ./functions.sh: No such file or directory
firmware-util.sh: line 53: prelim_setup: command not found

Thank you in advance

MrChromebox, 2021-05-15:
running on what device? under what OS?

time/date are correct on device?

Chrome-GitHub, 2021-05-15:
Good Afternoon Sir,

running on what device?
- Dell Chromebook 11 i3-4005u, Model CB1C13

under what OS?
- Fresh install of Ubuntu 20.04 LTS

time/date are correct on device?
- Yes sir, confirmed time/date are correct at the top of the default Ubuntu screen.

Thank you again for taking the time to help

MrChromebox, 2021-05-15:
Just booted Ubuntu 20.04.2 live ISO on WOLF, and no problems running the script from there. The errors you posted indicate something prevented the script from downloading the additional files it needs to run. Hard to say what the root cause is, but it's not a problem with the script or guithub's servers best I can tell

Chrome-GitHub, 2021-05-17:
Good Evening Sir,

Closing the loop on this issue. I am extremely grateful for both your work to help the greater ChromeOS community and taking the time to look into this matter. I confirm the script does work and it was a matter of the linux distro preventing the script from running correctly.

To help others who may experience the same difficulties I did, here are the distros that did not work for me out of the box:

- Puppy Linux - FossalPup64 9.5, error: need to add "iomem=relax" to kernal
- Ubuntu LiveCD - 20.04.2.0, error: could not install curl to run the command line commands?
- Ubuntu full install (including update & upgrade) - 20.04.2.0, error: could not download the supporting files (see original post)
- Slax - 9.11.0 64-bit, error: wouldn't boot up? I'm guessing it's because Rufus couldn't make the live USB with GPT partition scheme

Finally, I got it the script to work with a full debian install, after adding the user to the sudo group, installed curl, and modifying the grub configuration file to include "iomem=relaxed". Then I was able to successfully run the script and update the firmware.

Anways, that was a lot of words to say: "Thank you again for the wonderful work you're doing!"

MrChromebox, 2021-05-17:
still baffled as to why you had an issue on Ubuntu 20.04 live USB when I had none, but glad you got it sorted

closed 2021-05-17 by MrChromebox

ChromeOS-IS-NOT-COOL, 2023-01-31:
> running on what device? under what OS?
> 
> time/date are correct on device?

Yeah Not For me on ubuntu

MrChromebox, 2023-01-31:
> > running on what device? under what OS?
> > time/date are correct on device?
> 
> Yeah Not For me on ubuntu

what exactly am I supposed to make of this comment?

muhaabaminu, 2023-12-22:
I have asus c200 chromebook when i tried to install firmware utility using mrchromebox.tech command it keep "curl: (28) operation time out after 300448 milliseconds" after taking long time and i have a good internet connection please help