MrChromebox/scripts #155: Newer CloudReady (and perhaps other Chromium distributions) don't allow flashrom access even with `iomem=relaxed`

dragon788, 2021-05-28:
It appears that it might be an artifact of using a much newer kernel than most Chromebooks, and there are some new restrictions in the 5.4.x and newer series against root accessing certain devices like the EEPROM chips without enabling certain options, and unloading the Intel module as suggested on the Flashrom FAQs failed because it was build into the kernel NOT as a module, and even adding `iomem=relaxed` didn't resolve the error I was getting so I ended up having to boot a Linux LiveUSB instead to restore the stock firmware to a Chromebook I was testing a couple things with.

https://flashrom.org/FAQ

I've been looking around and I think I might have found a way to turn off the `kernel_lockdown` feature that prevents your scripts from working on a normal "secure" CloudReady installation if a user is planning on restoring the stock firmware and abandoning their CloudReady since the workaround requires disabling `verity` and making some other modifications that probably aren't advisable to run long term if the system is used for banking or other sensitive information.

MrChromebox, 2021-05-29:
I don't have any interest really in working around the stuff Cloudready does differently from ChromeOS. It's easy enough to boot another Linux live USB and update from there, IMO

dragon788, 2021-05-29:
Now that Neverware is owned by Google I'd imagine things will probably be more consistent between ChromeOS and CloudReady, and the kernel lockdown is probably something they are testing in CloudReady before rolling out to Chromebooks when they start shipping with 5.x kernels.

I made a couple changes that don't appear to have done much, but I'll continue to poke at it a little bit more since I'll be demonstrating the full RW_LEGACY and full UEFI options and installing Linux/CloudReady then reverting it ChromeOS to a group of students in a month or so. It would be nice to avoid making an extra dozen USB drives with Linux to go with the ChromeOS recovery drives and CloudReady installers.

I'll have to try enabling PXE for an install one of these days so I can boot Linux from my travel router instead of a flash drive.

MrChromebox, 2021-05-29:
>  the kernel lockdown is probably something they are testing in CloudReady before rolling out to Chromebooks when they start shipping with 5.x kernels.

that would be problematic for flashing Chromebooks, since booting with adjusted kernel params isn't easily doable