MrChromebox/scripts #156: Getting 'unable to download cbfstool utility' due to insufficient space in `/tmp/boot`

dragon788, 2021-06-10:
Depending on how many times a device has been recovered the `/tmp/boot` may not have enough free space for flashrom and cbfstool. It may be good to detect any `/tmp/boot/recovery_logs.XXXXXXX` folders and suggest/offer to delete them to avoid ambiguous errors.

Originally I was thinking that there was an issue with the Chromebook being Enterprise enrolled or a new feature of the r90 ChromeOS update because I was getting permission denied writing to `/tmp/boot/util/` but I could write to `/tmp/` just fine. It finally took running `df -h` to notice that there wasn't any free space and then exploring `/tmp/boot` to discover the logs.

MrChromebox, 2021-06-10:
didn't realize logs were stored there. probably safe to rm -rf /tmp/boot/recovery*, eh?

dragon788, 2021-06-10:
Yeah they definitely should be safe to clean up, I'm not sure if they are from Powerwash cycles or only show up on systems that have been recovered from USB/SD. I noticed that both of the "open box" Canadian Dell Latitude 5300 2in1 systems I picked up this week had half a dozen or more of those folders in the EFI partition after I had switched them to Developer Mode to install the RW_LEGACY firmware, but oddly I only ran out of space on one of them.

The interesting thing on the 2nd system where I decided to look through the folders was that some of the recovery logs were showing a CPU model that was most definitely NOT the one in the system, so I don't know if they were a result of an imaging process that was copying from a base device that had the clutter or where exactly they came from.

closed 2021-10-01 by MrChromebox