MrChromebox/scripts #158: how to disable Fw WP

sharroj, 2021-08-29:
have successfully used legacy firmware option to allow booting from usb.  Then i was able to install gallium os on acer 15 cb5-571.

But after leaving the machine uncharged for a while, I was no longer able to boot.  Having read the supporting documentation I realise that I needed to set GBB flags.  So I have run the mrchromebook script again but all options except the first (install legacy firmware) are listed as WP: enabled.  The documentation says it is not recommended to change this manually but I can't see a way forward.  Any Help?

MrChromebox, 2021-08-30:
> The documentation says it is not recommended to change this manually but I can't see a way forward. Any Help?

as per https://mrchromebox.tech/#devices, your device uses a screw for hardware write protection, so simply remove it an re-run my script.

that said, you'd be better off using the UEFI Full ROM firmware as your device is past or nearing EOL

sharroj, 2021-08-30:
Sorted! Thanks very much. I guess I was hoping to avoid using a screwdriver!

closed 2021-08-30 by sharroj