MrChromebox/scripts #159: ARM64/AARCH64 Support

krome-3k, 2021-08-31:
Please support devices with the Mediatek MT8183C SoC. There are quite a few of these out there.
Thanks in advance.

MrChromebox, 2021-08-31:
support for what exactly?

krome-3k, 2021-08-31:
> support for what exactly?

I want to run your firmware utility script but i am on ARM64 chromebook. Please help.

MrChromebox, 2021-08-31:
but to do what? Legacy Boot mode doesn't exist on ARM devices, so there's no RW_LEGACY firmware. ARM devices typically don't use UEFI either. And you can boot Linux with the stock firmware. So I'm not sure what you want exactly

krome-3k, 2021-08-31:
> but to do what? Legacy Boot mode doesn't exist on ARM devices, so there's no RW_LEGACY firmware. ARM devices typically don't use UEFI either. And you can boot Linux with the stock firmware. So I'm not sure what you want exactly

I want to remove chrome os and install another distro.. maybe dual boot?

krome-3k, 2021-08-31:
> > but to do what? Legacy Boot mode doesn't exist on ARM devices, so there's no RW_LEGACY firmware. ARM devices typically don't use UEFI either. And you can boot Linux with the stock firmware. So I'm not sure what you want exactly
> 
> I want to remove chrome os and install another distro.. maybe dual boot? Please tell me how to.

MrChromebox, 2021-08-31:
that's not something you need my script for. How-to is very device specific, and not something I'm able to help with. Look at Arch ARM to see if your device is supported/how-to, or other distros for ARM Chromebooks

closed 2021-08-31 by MrChromebox

krome-3k, 2021-09-01:
Arch linux as per https://archlinuxarm.org/platforms/armv8/mediatek/acer-chromebook-r13 boots into a blank screen but is for MT8173. As per https://archlinuxarm.org/platforms/armv8/generic doesn't boot with a beep. Archbox https://github.com/lemniskett/archbox and archrome https://github.com/gsf/archrome run into errors. Ubuntu server ARM64 and Debian ARM64 isos don't boot with a beep. Please help.

MrChromebox, 2021-09-01:
I'm not sure what you think I can do - your device simply isn't supported by those distros. No custom firmware is going to magically change that.

krome-3k, 2021-09-01:
Is it possible to port any of them for my device? Maybe by building from source? I am noob coder but i can follow instructions and google bugs. Can you tell me how is hardware support added say to archlinux if i build from source? Or do i have it all wrong and it cant be done? Please guide me.

MrChromebox, 2021-09-01:
I'm not an expert on ARM ChromeOS devices or ARM for Linux, there's nothing I can do for you I'm afraid. I'd recommend you ask for help on the Arch forums/IRC etc.

krome-3k, 2021-09-01:
Okay, thanks anyway.

MrChromebox, 2025-11-17:
there is no such thing as legacy boot on an ARM platform, legacy BIOS does not exist.
While the hooks for AltFw might exist, there's no universal payload for ARM the way there is on x86.
I have my hands full with 300+ x86 devices, I'll let someone else handle ARM if they want

scorpio79-debug, 2025-11-19:
[image].   Look at this please you mentioned that arm64 don't have legacy boot but this does

scorpio79-debug, 2025-11-21:
I added one this is what I was just trying to tell you in the first place chrome OS developer website shows how to do it