MrChromebox/scripts #160: No boot on KASUMI

GT3CH1, 2021-09-06:
After flashing the full UEFI firmware for KASUMI (Acer 311, The AMD variant) the device never boots. I can see the backlight turning on for less than a second, and then immediately turning off. I have tried using the firmware-util script, and a ch341a programmer, with no luck getting the firmware to boot.

I am able to restore the stock firmware via the programmer.

MrChromebox, 2021-09-06:
If you have a Suzy-Q, can provide a debug build to try and troubleshoot, but short of that nothing I can do other than check that the config is consistent with the other StoneyRidge boards. I have a BARLA and LIARA here and they both work with the current 4.14 release firmware

GT3CH1, 2021-09-06:
I just found a "quirk"
After reflashing the stock firmware, and enabling legacy booting again,
I've ended up in the UEFI  firmware. So I guess it works but not as
intended?

MrChromebox, 2021-09-06:
that's not a quirk, Stoneyridge devices, like all other devices using the new alternative bootloader menu, use Tianocore for legacy boot instead of SeaBIOS

closed 2021-10-01 by MrChromebox