MrChromebox/scripts #162: Firmware utility script error

NellyJelly, 2021-09-19:
Hi, 
   Trying to return 2 chromebox´s  back to stock but keep getting this msg. I have a. HP and a Asus, have tried different varients of Linux tried making usb on Windows and Mac and have tried using different USB sticks. I get the same result. Need help as my Linux skills are very limited.

cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh

Dload Upload Total Spent Left Speed
0 0 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0
0 185 0 0 0 0 0 0 --:--:-- --:--:-- --:--:-- 0
100 1252 100 1252 0 0 2469 0 --:--:-- --:--:-- --:--:-- 2469

Downloading supporting files...
firmware-util.sh: line 45: ./sources.sh: No such file or directory
firmware-util.sh: line 46: ./firmware.sh: No such file or directory
firmware-util.sh: line 47: ./functions.sh: No such file or directory
firmware-util.sh: line 53: prelim_setup: command not found

MrChromebox, 2021-09-19:
sounds like it's an issue with your internet connection to my site's server, not much I can do about that unfortunately.

NellyJelly, 2021-09-19:
Ok, thanks for replying, will keep trying.

NellyJelly, 2021-09-20:
Hi Matt,
      For some strange reason I can ping your site from my computer but not from any of the Chromeboxes I have, what do you think could be stopping me from connecting? I then used my  phones hotspot on a Asus Chromebox , I can ping your site but still cant download the supporting files. Is there something | I should be looking at.

MrChromebox, 2021-09-20:
I have no idea what the issue could be, but it doesn't appear to be server-side

NellyJelly, 2021-09-27:
Hi,
  For anybody else who is having this problem, there seemed to be problem with the linux live usb install so I ran these 
   sudo apt update && sudo apt upgrade
   sudo apt install curl
   and then I ran the Utility script again and it worked everytime..

MrChromebox, 2021-10-01:
cool, sounds like a non-issue with the script then

closed 2021-10-01 by MrChromebox