MrChromebox/scripts #163: SSL certificate expired

bmshizz, 2021-09-30:
using the instructions I get an SSL error when downloading firmware-util.sh, this can be bypassed with a -k switch in curl but I encounter another error that I Also suspect is SSL related

Error downloading cbfstool; cannot proceed.

Unable to download cbfstool utility; cannot continue

similar resolution noted at https://github.com/MrChromebox/firmware/issues/100

MrChromebox, 2021-09-30:
using various web tools, I see no certificates for mrchromebox.tech are expired, nor for the github domains from which files are fetched. Whatever is causing the issue is not on the server side

DouglasThurman, 2021-09-30:
I get the same thing.  All worked fine last week.  I'm using multiple HP Chromebook 14-SMB.  ChromeOS 76.  I'm following the guide and when I enter developer mode then a terminal tab, then shell...I get 
crosh> shell
chronos@localhost / $ cd ~; curl -L -O http://mrchromebox.tech/firmware-util.sh; sudo bash firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (60) SSL certificate problem: certificate has expired
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
bash: firmware-util.sh: No such file or directory
chronos@localhost ~ $

MrChromebox, 2021-09-30:
are we sure that the certificates in ChromeOS 76 aren't the issue?

DouglasThurman, 2021-09-30:
Well, I added the -k to the script and I'm at the error first listed in this thread.  If it's not the server, how would I check to see if Google borked the v.76 certificates?

DouglasThurman, 2021-09-30:
The strange thing is I've performed the BIOS flash and upgraded 14 of the ChromeOS 76 devices so far and today is the first time I've come across errors.  I thought Google might have caught on to my loading Neverware's Cloudready ChromeOS onto them.

MrChromebox, 2021-09-30:
sure, but now that it's a different day and something is now expired, why would you assume it's the cert on the server (which you can check easily via googling) rather than the cert/CA on the very old OS you're running?

DouglasThurman, 2021-09-30:
Apologies then.

MrChromebox, 2021-09-30:
I mean, and easy test is can you download the file from your phone or another machine? If so, the issue isn't on the server side

DouglasThurman, 2021-09-30:
Okay I tried the process on an HP Chromebook 11 G4 which is using ChromeOS v.91.  I get the same errors with curl, and when I add the -k to it:
chronos@localhost ~ $ cd; curl -L -O -k mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0   185    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100  1252  100  1252    0     0    868      0  0:00:01  0:00:01 --:--:--   868

Downloading supporting files...
Error downloading cbfstool; cannot proceed.

Unable to download cbfstool utility; cannot continue

What am I missing?

DouglasThurman, 2021-09-30:
Used phone hotspot...same.  Tried running the script on a computer with ZorinOS 16 and the hotspot...same.  Could there be an issue with curl?

MrChromebox, 2021-09-30:
adding '-k' only helps with the initial script download, it doesn't carry through to the downloads the script makes, which is why it's failing still. I can't reproduce this here from my main Linux box, or any number of Chromebooks running ChromeOS. And I'm not familiar enough with SSL/curl to debug the issue without doing some research

danielktaylor, 2021-09-30:
I'm also getting this on my CN60 today, trying to install for the first time.

MrChromebox, 2021-09-30:
the CN60 is running what version of ChromeOS again? pretty old at this point?

I'm guessing I'll need to find a way to work around this. maybe just pass in -k via command line param

danielktaylor, 2021-10-01:
Forking your repo and adding -k to all the curl commands seems to resolve it for me. Certainly not the "right" way...

MrChromebox, 2021-10-01:
I'll just allow -k to passthru via cmd line

MrChromebox, 2021-10-01:
script and documentation updated with info on how to mitigate this

closed 2021-10-01 by MrChromebox

Itzdonskynze, 2022-09-24:
Having an issues

leftoverbeefcake, 2022-10-01:
I'm getting the exact same error as bmshizz earlier, cannot download the script on my Asus C202s, visiting https://mrchromebox.tech/ with Firefox on an iMac gives me the "warning:potential security risk ahead" page:

"Firefox detected an issue and did not continue to mrchromebox.tech. The website is either misconfigured or your computer clock is set to the wrong time.

It’s likely the website’s certificate is expired, which prevents Firefox from connecting securely. If you visit this site, attackers could try to steal information like your passwords, emails, or credit card details."