MrChromebox/scripts #166: cbfstools can't load

DouglasThurman, 2021-10-04:
Thank you @MrChromebox  for updating your script.  Now though I'm still getting the error:

Error downloading cbfstool; cannot proceed.
Unable to download cbfstool utility; cannot continue

This is after downloading and running the firmware-util.sh.  Is there a way to download the cbfstools separately and into the directory it's being called from?  

Apologies for wrong terminology I'm a Windows guy.

MrChromebox, 2021-10-04:
the issue does not occur when using the updated command listed on my site, to mitigate the certificate expiration error on older ChromeOS builds. Ensure you are running it exactly as per my website

DouglasThurman, 2021-10-04:
So your website shows two different install methods.  One for a shell in ChromeOS, and one for running it in Linux.  The ChromeOS instructions show three separate lines to enter in.  This is what happens when I do:

[Pro Tip] Use 'Open as Window' or 'Fullscreen' to prevent Ctrl-W from closing your terminal!
[Pro Tip] See [crosh] for more information.

Loading extra module: /usr/share/crosh/dev.d/50-crosh.sh
Welcome to crosh, the Chrome OS developer shell.

If you got here by mistake, don't panic!  Just close this tab and carry on.

Type 'help' for a list of commands.

If you want to customize the look/behavior, you can use the options page.
Load it by using the Ctrl+Shift+P keyboard shortcut.

crosh> shell
chronos@localhost / $ cd; curl -LO mrchromebox.tech/firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (60) SSL certificate problem: certificate has expired
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
chronos@localhost ~ $ sudo install -Dt /usr/local/bin -m 755 firmware-util.sh
chronos@localhost ~ $ sudo firmware-util.sh

Downloading supporting files...
Error downloading cbfstool; cannot proceed.

Unable to download cbfstool utility; cannot continue

DouglasThurman, 2021-10-04:
AHA!  The first line of script from the ChromeOS instructions doesn't have the -k added.  Added it and now the whole thing clicks.  Yay!

closed 2021-10-05 by MrChromebox