MrChromebox/scripts #168: Asus cx5400 FW installs tianocore

wbskinner, 2021-11-25:
I have an Asus cx5400, I ran the firmware script, picked #1 option, rebooted, in the altfw menu - tianocore is now the only bootloader listed, (replaced the previous u-boot item).  After selecting it, the system is frozen, only option is to poweroff using power button on the laptop. I expected to see the seabios menu ( I have older chromebooks that I have used the firmware script on).

MrChromebox, 2021-11-26:
ChromeOS devices which utilize the Alternative Bootloader Menu have Tianocore as the RW_LEGACY payload, not SeaBIOS. This is normal/expected behavior. If Tianocore doesn't boot on your device, it's due to bugs in the factory firmware and not much I can do. It does work on other Tigerlake-based Chromebooks.

closed 2021-11-26 by MrChromebox

wbskinner, 2021-11-26:
Thanks for replying so soon, I'm going to probably dig deeper into  whats going on - maybe talk to Asus, if I get it  to work I will let you know.

MrChromebox, 2021-11-26:
Asus won't know anything, Google is the one responsible for the firmware. And Developer Mode /Altfw are unsupported options/features