MrChromebox/scripts #172: Randomly get no display until hard reboot w/ ASUS C302

flyoffacliff, 2021-12-14:
I've been having issues with an ASUS C302 (Google Cave) Chrome book. Sometimes I randomly get a black screen. Isle city backlight is off, but power light is on. the only way to fix it is to do a hard shutdown by holding the power button for a while and then wants to light goes off wait a minute and then hold it a couple more seconds and hope I get lucky and it boots back up (it often takes a few tries).
This was with the latest version of Gallium OS on RW_LEGACY mode. The issue does not occur when booting into Chrome OS. I thought it was a bootloader issue and not a gallium OS issue because it never even made it to the developer options screen. (I don't think Ubuntu worked right either).
Then I switched to the full ROM firmware (build 3.14 that was just released in July), and even after installing the updates it seems to be doing the same thing.

MrChromebox, 2021-12-14:
this might be a hardware issue given you're the first to report it and the device is 5+ years I've been supporting it

MrChromebox, 2021-12-14:
also, closing as this isn't an issue with my script

MrChromebox, 2023-02-04:
not a script issue; closing

closed 2023-02-04 by MrChromebox