MrChromebox/scripts #173: The only way to select a boot device is rebooting throughout the script

k2arim99, 2022-01-02:
Title, my device is a 2013 pixel link, is this intended behavior?, Had no issues with the installation

MrChromebox, 2022-01-02:
I have no idea what you are trying to say

k2arim99, 2022-01-02:
When I reboot and control L to legacy mode I get directly to "booting from the hard drive", when I reboot from the firmware-util.sh script I get to choose the boot drive correctly

MrChromebox, 2022-01-03:
that has nothing whatsoever to do with my script, it's simply SeaBIOS being buggy / behaving differently on a cold vs warm reboot.

that said, LINK is well past EOL, and you'd be much better served removing the firmware WP screw and flashing my UEFI Firmware instead of screwing around with RW_LEGACY/SeaBIOS

closed 2022-01-03 by MrChromebox

k2arim99, 2022-01-03:
That's helpful thanks!