MrChromebox/scripts #177: ChromeOS Firmware Utility Error

ascii1b-pixel, 2022-01-23:
I have an ASUS CN60 Chromebox which has Ubuntu 20.04.3 installed. The time and date are correct.

I haven't used it in a while and decided to run the firmware updater. Here's what happened:

$ cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0   185    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  1441  100  1441    0     0   1388      0  0:00:01  0:00:01 --:--:-- 24016
[sudo] password for ascii27: 

Downloading supporting files...

firmware-util.sh: line 52: ./sources.sh: No such file or directory
firmware-util.sh: line 53: ./firmware.sh: No such file or directory
firmware-util.sh: line 54: ./functions.sh: No such file or directory
firmware-util.sh: line 60: prelim_setup: command not found

I read that this error was already reported about a year ago and I tried to follow the poster's suggestions, but nothing worked for me.

I'm always so impressed with how helpful MrChromebox is in replying to user's questions, so I'm hoping he can figure this out.
Thanks so much

MrChromebox, 2022-01-24:
I don't have any suggestions offhand, other than to boot a newer Ubuntu USB and try from there

ascii1b-pixel, 2022-01-24:
Yep. It was definitely something to do with Ubuntu. Even though it was up to date it still didn't work.
I've been wanting to try some different "flavors" and I loaded Zorin and amazingly your script worked worked like it's supposed to. I'm up to date now.
Thanks so much for responding.

closed 2022-01-24 by MrChromebox