MrChromebox/scripts #179: firmware-util.sh script fails to detect hardware platforms for certain chromebooks

TheMadman, 2022-01-24:
Running the firmware-util.sh script on a Lenovo S345-14 (AMD A6) with the hardware ID LIARA produces the following output on the firmware menu:

```
**   Device:  (GRUNT)
** Platform: (unrecognized)
```

This is from running the script inside of a custom Linux install, booted in RW_LEGACY mode.

The firmware still installs and allows booting into legacy mode.

MrChromebox, 2022-02-07:
sadly for many ChromeOS devices, there is no way to disambiguate them when booted in legacy boot mode, since depthcharge (the chromeOS firmware payload) fails to populate the HWID into the SMBIOS tables. I suppose I could try to read it directly from the GBB firmware region...

MrChromebox, 2022-02-08:
should be fixed now - verified working on my LIARA running Linux w/RW_LEGACY firmware

closed 2022-02-08 by MrChromebox