MrChromebox/scripts #180: Asus Chromebox CN62 No Boot After Full Firmware Install

HelgenX, 2022-02-10:
I know this is an old thread, but I wanted to add to this that I installed mrchromebox on a Asus chromebox C062, and it won't read any os I put on it, it won't do anything at all. Tried it with a USB 2 and 3 drive, tried Ubuntu, windows 8.1, mint, older versions of those os's as well. Just a black screen. I also did not choose backup, but I think even if I did, since nothing posts, it wouldn't matter.

Only thing I haven't tried was using dp port to see if anything at all posts, but as soon as I press power, it turns off... Which could also mean video problems,maybe hdmi not read on post?

Edit: tried display port, nothing. It's completely bricked. All I did was install it to overwrite firmware, it won't do anything

MrChromebox, 2022-02-11:
https://wiki.mrchromebox.tech/Unbricking

MrChromebox, 2022-02-11:
without examining the image extracted from your device, I can't say precisely either, but there's a reason the flashing script has a warning

HelgenX, 2022-02-13:
Thanks for the response. The chromebox I had was just an experiment I got for free, but I now know that anything chrome Os is a plague. I got a NUC, problem solved.

closed 2022-02-13 by HelgenX

MrChromebox, 2022-02-13:
there's nothing wrong with ChromeOS or the box, you just got a bad firmware flash, which happens on occasion when going from ChromeOS firmware to custom. It's easily rectified with $10 in hardware