MrChromebox/scripts #183: Asus Chromebox 3, no fan control on Ubuntu?

bam80, 2022-03-02:
Hi,
didn't notice fan speed variation on usual tasks - system update, etc.
Does fan control work on Ubuntu by default?
If not, any hints how to set it up?

MrChromebox, 2022-03-02:
fan control on 7th/8th-gen Chromeboxes is just like Chromebooks - handled via the EC, and controls are not exposed to the OS

you would need to compile Google's ectool in order to tweak anything fan-related

bam80, 2022-03-02:
Hmm, it's strange the EC didn't change fan speed during system update - on ChromeOS it seemed to be more sensitive. But that might be just my perception..
Thanks, I probably need to give it more time - maybe EC control happens to be satisfyable..

closed 2022-03-02 by bam80