MrChromebox/scripts #184: firmware-util.sh: line 52: ./sources.sh: No such file or directory

bam80, 2022-03-03:
```
$ cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0   185    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100  1441  100  1441    0     0    811      0  0:00:01  0:00:01 --:--:--   811

Downloading supporting files...
firmware-util.sh: line 52: ./sources.sh: No such file or directory
firmware-util.sh: line 53: ./firmware.sh: No such file or directory
firmware-util.sh: line 54: ./functions.sh: No such file or directory
firmware-util.sh: line 60: prelim_setup: command not found
```
For some reason curl can't create the files being run as root:
```
$ sudo curl -LOv https://raw.githubusercontent.com/MrChromebox/scripts/master/functions.sh
...
Warning: Failed to create the file functions.sh: Permission denied
* Failure writing output to destination
...
curl: (23) Failure writing output to destination

MrChromebox, 2022-03-03:
boot a live USB with a sane version of curl? I can adjust to use TMP=mktmp() or something I guess

bam80, 2022-03-03:
As workaround I reinstalled curl with apt.
Setting TMP won't probably help as it seems still interpreted within that special /tmp/snap.curl/ directory..

PSCGOhio, 2022-07-07:
I;m also getting that same error, but it gets a 400 error trying to source the 3 files?  I manually got them, but the way  Ubuntu is set up with 22.04 on my Dell Chromebook I can't get perms to copy them where they belong.  And I can't get into recovery mode nor update from SeaBIOS to fix that.  Ideas?  Can the script be rewritten to find those 3 where I have them?  symbiotic link or something?

bam80, 2022-07-07:
> As workaround I reinstalled curl with apt.

@PSCGOhio did you try that?

PSCGOhio, 2022-10-11:
Yep, if it was in the thread it was tried.  The same obstacle of no SU
abilities stops me cold.

MrChromebox, 2022-10-11:
@PSCGOhio what are you trying to do? There is no Recovery Mode or SeaBIOS with my UEFI firmware. snap-installed curl is problematic, install via apt instead. You should have no problem then

cutervino, 2023-06-05:
Problem solved: I installed Ubuntu 23.04 on my Asus Chromebook C302C. To make a long story short, I managed to installed Mrchromebox. but no sound (Dummy sound), Bluetooth not working, etc. So I decided to install Qube OS, and it failed, leaving my Chromebook bricked. I found at medium.com/techtalkers, the solution using Ubuntu live usb. It failed with 22.04.2 (No curl), it was a mess: freezing, screen up side down, etc. Then I tried with 23.04 and I had the same problem Bam80 describes. But I didn't have the problems of 22.04.2. It was kind of neat, so when it failed, just out of frustration, I hit Install Ubuntu 23.04 instead of try. And it was lightning fast. It updated neatly, I installed immediately Nordvpn, VLC.  So far the only problem is no internal speakers sound, (Bluetooth works) and no tablet mode, but that's nothing compared with a bricked Chromebook and happy to say goodbye to Google consumerism. All this started because Google announced no more support starting some fatidic day of June 2023. I'm getting rid of all Google services, just to save the world.

cutervino, 2023-06-05:
Ah, I forgot to state that the freaking rabbit still appears at the beginning of booting

MrChromebox, 2023-06-05:
@cutervino my FAQ is pretty clear about needing to use apt-installed curl, not snap-installed, on newer versions of Ubuntu . There's simply no way to work around the limitations of snap-installed curl.

> Ah, I forgot to state that the freaking rabbit still appears at the beginning of booting

I'm not sure I understand the problem, that's the firmware boot splash screen

cutervino, 2023-06-05:
I apologize MrChromebox, I shouldn't have used that word. No problem at all. I'm neither a geek, nor a techy guy. I'm just a citizen of the world that wants to help with the climate change. It's just that, when Ubuntu asked me to erase the whole disk at installation, I thought it'll delete everything. We all know your hard work on this project, and we thank you for that. 
So I take this opportunity to ask you if there's a way to get the sound working and if it's possible to get tablet mode. I appreciate your help.

MrChromebox, 2023-06-05:
> It's just that, when Ubuntu asked me to erase the whole disk at installation, I thought it'll delete everything

yes, it will erase everything on the target installation disk. The firmware is not stored there :)

>  if there's a way to get the sound working and if it's possible to get tablet mode. I appreciate your help.

on CAVE? Yes, but not using Ubuntu. Check the links on my FAQ page, as well as the #linux channel on the chrultrabook Discord server.

closed 2023-10-14 by MrChromebox