MrChromebox/scripts #185: Fw WP: Disabled with WP screw in place

bam80, 2022-03-03:
I put WP screw back in place, but still the script reports me WP is disabled.
How does it work actually?

MrChromebox, 2022-03-03:
https://wiki.mrchromebox.tech/Firmware_Write_Protect

the screw (HW) only enforces the software WP settings, which are disabled when you flash the UEFI firmware via my script

bam80, 2022-03-04:
https://kodi.wiki/view/Archive:Chromebox#Disable_Firmware_Write_Protect:
> Important: The write-protect screw should be left out permanently. 

Why is that?

MrChromebox, 2022-03-04:
because on some platforms, having hardware WP enabled with software WP disabled causes weird behavior

bam80, 2022-03-04:
Strange, in my case it's opposite - I can't even boot my PD-powered system without the screw (original fw).
The worst thing I can't boot it at all with UEFI fw: #186

MrChromebox, 2022-03-04:
correlation != causation. The UEFI firmware updates the EC firmware, which handles PD as well. So something in the EC update broke PD.

bam80, 2022-03-04:
Oh, good to know! Is it necessary to update the EC fw? I'm trying to find a way we could bisect the problem.

MrChromebox, 2022-03-04:
I'll have to do that here, but first figure out how I can do PD and get ccd serial access simultaneously

bam80, 2022-03-04:
Servo board seem does have that capability:
https://chromium.googlesource.com/chromiumos/third_party/hdctools/+/main/docs/servo_v4.md#type_c-version

Anyway, if that sort of debugging doesn't currently possible, could we just build test UEFI fw without EC fw, or with stock EC fw?

MrChromebox, 2022-03-04:
I have a servo v4, just lent it out unfortunately. I can revert to the stock/RO EC firmware and bisect from there, just a question of time

bam80, 2022-03-04:
I would be very gratitude. Thanks.

MrChromebox, 2022-03-05:
drop me an email for a test build, will see if it fixes the issue for you. mrchromebox at gmail

bam80, 2022-03-08:
> correlation != causation. The UEFI firmware updates the EC firmware, which handles PD as well. So something in the EC update broke PD.

After applying a workaround for PD boot up by restoring Google's EC RW fw and removing EC RW fw from UEFI fw to disable swsync, the box still boots with WP screw in place only, so this line on [Wiki](https://kodi.wiki/view/Archive:Chromebox#Disable_Firmware_Write_Protect:
) should probably be corrected:
> Important: The write-protect screw should be left out permanently.

MrChromebox, 2022-03-08:
> After applying a workaround for PD boot up by restoring Google's EC RW fw and removing EC RW fw from UEFI fw to disable swsync, the box still boots with WP screw in place only, so this line on [...]

I'm still not sure what to make of this issue. I am able to boot via PD on my CB3 with the WP screw removed using my latest ECRW update. Further, if the EC gets reset and boots in RO mode, it doesn't jump to RW until relatively late in the AP booting process, at which point PD negotiation is long done.

bam80, 2022-03-08:
You said you reproduced it once, can you try it again?
Anyway, until it sorted, we could place a link to this issue on the Wiki. Otherwise, as we see consequences might be pretty dramatic.

MrChromebox, 2022-03-08:
I just "updated" to ECRW v1.1.0-7903-g93805b9b52on my box here, no issues with PD, WP removed.

bam80, 2022-03-08:
> Further, if the EC gets reset and boots in RO mode, it doesn't jump to RW until relatively late in the AP booting process, at which point PD negotiation is long done.

Please keep in mind there are actually two issues:
- bootloop
- not being powering at all with blinking red LED

The bootloop passes PD negotiation, obviously.
Still I experience it even on stock fw with 100% reproducibility.
So it's not an issue with your fw, it's an issue with Wiki leading to a wrong path.

MrChromebox, 2022-03-08:
I had someone check the schematics, the WP screw isn't in any way tied to PD. I have no idea why you're seeing the results you are, but I can't reproduce them

bam80, 2022-03-08:
We still have different PD sources.
Mine is 27" DELL U2722DE display with 90W PD.

My guess WP screw might be not tied to PD schematically but rather logically.

bam80, 2022-03-08:
> Anyway, until it sorted, we could place a link to this issue on the Wiki. Otherwise, as we can see consequences might be pretty dramatic.

Meanwhile, I made a note on the Wiki. If you can make it better, please do:
https://kodi.wiki/index.php?title=Archive%3AChromebox&type=revision&diff=241698&oldid=235000

closed 2022-03-08 by bam80