MrChromebox/scripts #186: Asus Chromebox 3: can't boot from USB-C Power Delivery after flashing UEFI firmware

bam80, 2022-03-03:
First of all it still boots from power adapter.
Then, if I swap to PD with WP screw in place, LED blinking red and it doesn't boot.
If I remove the screw, it bootloops.

With original firmware, I can boot from PD with screw in place only. If I remove the screw, get bootloop.

So it seems no way of using native Linux with PD currently, which is a whole point I bought the device.

Any help? @MrChromebox could you try to reproduce, please?

bam80, 2022-03-05:
> With original firmware, I can boot from PD with screw in place only. If I remove the screw, get bootloop (need to detach from power before removing the screw, otherwise it can still boot).

This particular issue is reported here:
https://issuetracker.google.com/issues/222675683

MrChromebox, 2022-09-26:
closing as this isn't a script issue

closed 2022-09-26 by MrChromebox

bam80, 2023-06-29:
Could we move the issue between the projects maybe?
This should be placed in https://github.com/MrChromebox/firmware/issues probably.

First needs to be reopened:
https://docs.github.com/en/issues/tracking-your-work-with-issues/transferring-an-issue-to-another-repository

bam80, 2024-03-16:
> Could we move the issue between projects?

@MrChromebox