MrChromebox/scripts #187: Suspend doesn't work with UEFI fw

bam80, 2022-03-07:
Ubuntu 21.10, Asus Chromebox 3.
On suspend, display blanks but fan still active. LED lit as usual.
There is no way to "wake" the system - it doesn't react to input and power button any more.

MrChromebox, 2022-03-07:
No such issue on Pop_OS! 21.10 on my CB3

bam80, 2022-03-07:
That was with test fw you sent me, I'll retest with latest one

bam80, 2022-03-07:
Do we still have our EC RO versions different?

bam80, 2022-03-07:
I also have FW protect screw in place. Can it impact?

MrChromebox, 2022-03-07:
it shouldn't

bam80, 2022-03-07:
Well I could test Pop_OS! too (I tested from Live image)

bam80, 2022-03-07:
With Pop_OS! Live got somewhat different result: system does wake up on keyboard press now.
Still, it doesn't sleep actually: fan continues to run and LED lit constantly.

@MrChromebox does system actually sleep on your side - fan, LED, etc.?

bam80, 2022-03-10:
> @MrChromebox does system actually sleep on your side - fan stops, LED blinks, etc.?

@MrChromebox ping.

MrChromebox, 2022-03-10:
depends on the type of suspend used. With the default s2/idle, no. with s3/deep, yes

bam80, 2022-03-10:
I did `systemctl suspend`, I suppose it's s3.
So our results differ. Do you have idea why?
How could I debug it further?

MrChromebox, 2022-03-10:
don't assume, verify: `cat /sys/power/mem_sleep` will show options available, with the current one being bracketed. 

Min looks like:
`$ cat /sys/power/mem_sleep` 
`s2idle [deep]`

bam80, 2022-03-10:
Yohoo! Mine was `s2idle`. One problem less :-D

Did you do additional tuning of the system? I just booted Pop_OS! 21.10 Live. I've never faced s2/idle is default before. Maybe after installing system it's default is changed, will see..

Many thanks.

closed 2022-03-10 by bam80

MrChromebox, 2022-03-10:
seems to be the default on newer kernels? I'd have to check and see if anything changed with coreboot in terms of low power modes advertised

bam80, 2022-10-12:
Any progress since then?
It seems people continue complain about this `s2idle`, does it actually work for this platform to be advertised in "FADT" (whatever it means)?
https://github.com/systemd/systemd/issues/13451#issuecomment-1274364914

If it _should_ work in theory, probably we shouldn't remove it from FADT, but then the question is why it doesn't work on practice?