MrChromebox/scripts #188: Procedure to override EC RW firmware

bam80, 2022-03-07:
Original firmware had a feature to "Preventing the RW EC firmware from being overwritten by Software Sync at boot" via GBB (Google Binary Block) flags:
https://review.coreboot.org/plugins/gitiles/chrome-ec/+/fd9d07de6d6f688f80e489d67977ad1070a08a15/README.md#Preventing-the-RW-EC-firmware-from-being-overwritten-by-Software-Sync-at-boot

Now we don't have GBB (but Software Sync was implemented), so the procedure to override EC RW fw without rebuilding the whole UEFI firmware would be helpful.
https://github.com/MrChromebox/coreboot/commit/f6f02eef9f33103e9d101e2be33edaac54e30e93#commitcomment-68158449:
> while I could make it easier, rebuilding isn't needed - you can just remove the ecrw file and hash from cbfs and flash back

I think instructions how to do that here or on the Wiki would be useful.

MrChromebox, 2022-03-07:
this isn't an issue with the script or script functionality; please re-open under firmware instead (and that's not a promise to implement)

closed 2022-03-07 by MrChromebox

bam80, 2022-03-07:
https://github.com/MrChromebox/firmware/issues/335