MrChromebox/scripts #190: support for ambary lake asus c433

coutouly, 2022-03-16:
every thing goes rigth
but when i bootup i get this error
seabios ...
booing from hard disk ...
sislinux 6.03 edd load error - boot error

so i dont see my usb disk . no list
can you fix it ?
thanks

coutouly, 2022-03-16:
> > every thing goes rigth
> 
> without context, I have no idea what this means
> 
> > can you fix it ?
> 
> https://mrchromebox.tech/#faq
> 
> closing as this isn't an issue with my scripts

the script says that the job is ok 
but when i boot up it doesnt see the usb disk so i canot bought on it.

coutouly, 2022-03-16:
could you help me ?

MrChromebox, 2022-03-16:
my FAQ addresses the issue. There's is nothing more that I can do to help, you simply need to try another USB stick or another port.