MrChromebox/scripts #191: ChromeEC: No valid hash (status=0 size=2058478072). Compute one...

bam80, 2022-03-17:
This string is always present in `cbmem` log, so hash is not cached which creates additional delays and potentially problematic: #186 
```
Google Chrome EC: Initializing
EC returned error result code 1
Google Chrome EC: version:
        ro: fizz_v1.1.7899-258e4a3d2
        rw: fizz_v1.1.7909-6fc5ebe65
  running image: 1
CBFS: Found 'ecrw.hash' @0x9d200 size 0x20 in mcache @0x7abfd214
ChromeEC SW Sync: Expected hash: c5a1127f71d172f246e943dfb5487f1cac288a23d1aee75946c1e34fb2df16ff
ChromeEC: Getting hash:
ChromeEC: Getting hash:
ChromeEC: Getting hash:
ChromeEC: No valid hash (status=0 size=2058478072). Compute one...
ChromeEC: Starting EC hash:
ChromeEC SW Sync: current EC_RW hash: c5a1127f71d172f246e943dfb5487f1cac288a23d1aee75946c1e34fb2df16ff
ChromeEC SW Sync: EC_RW is up to date
Google Chrome EC: version:
        ro: fizz_v1.1.7899-258e4a3d2
        rw: fizz_v1.1.7909-6fc5ebe65
  running image: 2
PNP: 0c09.0 init finished in 989 msecs

MrChromebox, 2022-03-18:
I'm not sure why you think this is a problem, but it's not. The delay to calculate a hash on a normal boot is sub 5ms. And the issues that you're seeing on your device w/r/t USB-PD are not only not reproducible by me, but also not reported by any other Fizz users.

bam80, 2022-03-18:
I don't think this is necessary the problem, but it might be, directly or indirectly.
Anyway, as I understood you this is not expected behavior, as it should cache and compute the hash only once.

The fact you can't reproduce it doesn't tell us much, I guess.
Do we know many users with my use-case? With my display model?
But we know a user on reddit not being able to boot with PD, also. IIRC, there were a few such reports.

MrChromebox, 2022-03-18:
> Anyway, as I understood you this is not expected behavior, as it should cache and compute the hash only once.

it seems Fizz is different with using multiple RW ec fw options.

> The fact you can't reproduce it doesn't tell us much, I guess.

the log is the same as mine; I can't reproduce your USB-PD issues

MrChromebox, 2022-09-26:
closing as this isn't a script issue

closed 2022-09-26 by MrChromebox