MrChromebox/scripts #193: volteer support?

lanrat, 2022-04-13:
Is the "volteer" board supported?

I don't see its firmware option listed here: https://github.com/MrChromebox/scripts/blob/master/sources.sh

But it does look like you [explicitly called it out as supported on reddit](https://www.reddit.com/r/coreboot/comments/st5125/comment/hx1umv9/?utm_source=reddit&utm_medium=web2x&context=3).

Thanks.

MrChromebox, 2022-04-13:
volteer is the baseboard for all TGL Chromebooks, for which I only have RW_LEGACY support currently. Nothing preventing me from releasing UEFI firmware other than time and testers.

closed 2022-04-13 by MrChromebox

lanrat, 2022-04-13:
I'm happy to help test if needed.

For RW_LEGACY, where can I find information on using this? I'm somehow failing to find the correct results when searching.

MrChromebox, 2022-04-13:
not sure what you mean by "results"

lanrat, 2022-04-13:
How I can flash the RW_LEGACY formware to boot linux.

MrChromebox, 2022-04-13:
using my script. If your device isn't recognized properly, let me know the board name / HWID and I'll add/fix it

gitguest0, 2022-07-25:
I would be interested in testing as well, i have a lillipup/volteer (Flex 5 13ITL6) device.
I should be able to unbrick the device with a modified SOIC8 clip if that would be necessary.

[image]