MrChromebox/scripts #196: ASUS CN60 Panther UEFI firmware reporting checksum fail

rosera, 2022-04-14:
Machine: ASUS CN60

Downloaded the latest script and using the UEFI option Full Rom.

Looks to be an error when Downloading `coreboot_tiano-panther-mrchromebox_220220409.rom`

Message: `Firmware download checksum fail: download corrupted, cannot flash.`

Can someone confirm if there is an issue with the firmware or another issue?

MrChromebox, 2022-04-14:
no problem updating my panther box here, so the corrupted download is likely a momentary server glitch

closed 2022-04-14 by MrChromebox

rosera, 2022-04-14:
I compared the files sizes between different machines and the rom download was smaller than the expected 8.0M on the target machine. I ended up zipping the rom and sha on a second machine, then amending the script to unzip the fixed archive. Then perform the sha1 comparison on the target device, which would now pass successfully. BIOS updated successfully.

Many thanks for the quick response.