MrChromebox/scripts #199: Problems with AMD CPUs?

fractalf, 2022-04-27:
Hi,
I've read a couple of other issues here mentioning some earlier problems with AMD CPUs.

I just bough a **Lenovo 300e Chromebook 2nd Gen AST (AMD A4-9120C RADEON R4)** with the sole purpose of running Linux on it.

I've tried installing galliumos and ubuntu with *chrx* and Updating the RW_LEGACY firmware with the **firmware-util-sh** script, but nothing seems to be able to boot into anything except ChromeOS, which ofc is completely useless.

So, did I just waste 200$?

Thank you

MrChromebox, 2022-04-28:
it's 50-50 if RW_LEGACY works on StoneyRidge devices. GalliumOS doesn't support Stoney at all. UEFI Full ROM firmware will boot, but expect GPU and audio to be problematic, among other things. Stoney just isn't a good platform for running Linux.

fractalf, 2022-05-02:
Ok, thanks a lot for explaining :) 

..I'm sending this laptop back to the shop for å refund

closed 2022-05-02 by fractalf