MrChromebox/scripts #200: pixbook2017 Firmware download checksum fail

jinigithub, 2022-04-29:
I'm using the MrChromebox script to install Cloudready on my immaculate but 'end-of-life' pixelbook2017.
It starts downloading the Full ROM firmware but after a few minutes gives a red-text message:
Firmware download checksum fail; download corrupted, cannot flash.

i tried several type of network env but ran to the same error

jinigithub, 2022-04-29:
then i found the script can't recognize my device,which caused checksum fail,may i force it to falshed?
some time ago,something was wrong with the pixelbook and i reinstalled os with some warning,but used it normal

MrChromebox, 2022-04-29:
>then i found the script can't recognize my device,which caused checksum fail,may i force it to falshed?

what? post a pic of the script menu

MrChromebox, 2022-04-29:
Also, CloudReady/CrOS Flex don't support the Pixelbook. And it's not EOL yet.

jinigithub, 2022-04-30:
yes, it was I had a wrong hwid,set one and resolved,tks for reply.

closed 2022-04-30 by MrChromebox