MrChromebox/scripts #201: pixebook with platform unrecognized

jinigithub, 2022-04-29:
hi,pixelbook 2017,have disabled WP,but showed platform unrecognized and can't activate install uefi menu
is there any solution of mark platform intel kabylake to the script?

MrChromebox, 2022-04-29:
Pic of menu

jinigithub, 2022-04-30:
it resulted I had a wrong hwid,i searched new hwid from web and tried 2times,done.
but a new problem occurred,i installed win10/win11,everything is okay except display rotation(instead in software adjustment) and keyboard.
windows10/11 had recognized it as pixelbook keyboard and said driver okay,but none of keys is valid.
while I installed manjaro,the keyboard is ok.

MrChromebox, 2022-04-30:
I don't support running Windows on any ChromeOS devices, that's what r/chrultrabook is for

closed 2022-04-30 by MrChromebox