MrChromebox/scripts #202: U-Boot/Depthcharge as payload with coreboot

EMLommers, 2022-05-06:
Hi Matt,
is it possible to add u-boat/depthcharge as coreboot payload?
after switching to the firmware we can still use the same bootable chromeos kernels.
Can  you look into this? thinbootloader has it also as payload.. and more.
Thanks
Ernst

MrChromebox, 2022-05-06:
one can certainly build upstream coreboot with depthcharge as a payload, but I'm not sure why you'd want to offhand. Changing the payload on one of my builds isn't worth the effort when you can just clone my repo and rebuild whatever config you want

EMLommers, 2022-05-06:
It will not be a standard selectable  payload for coreboot? And select the partition of kernel.
it is conveneient for people  who use original / modified   chromeos  kernel and normal linux distribution.
Do you think it is not something you can add in the firmware. People who use original chromeos,  cb-linux/breath or hexdump's imagebuilder and thinking about chnaging firmware can still keep their kernels, system.

MrChromebox, 2022-09-26:
closing as this isn't a script issue

closed 2022-09-26 by MrChromebox