MrChromebox/scripts #203: Intel ME on Lenovo thinkpad 13?

that0nepers0n, 2022-05-08:
Could there be intel me enabled in the firmware for this machine because I’m pretty sure that’s why I get a gpu error whenever I hackintosh anything above Mojave on my chrome book and I’ve googled for a solution but there’s only a few about chrome books and they all have the same problem with Catalina and others above Mojave

MrChromebox, 2022-05-08:
on most ChromeOS devices, the ME is hidden from the OS at boot. I'd have to do a custom build to enable it, so something that's best done by the user. Cloning the coreboot repo and compiling yourself isn't all that hard

that0nepers0n, 2022-05-08:
Ah ok I have no experience with compiling coreboot would you point me in the right direction and how would I go about enabling it? And also that you for all that you do in the community!

MrChromebox, 2022-05-08:
what is the ChromeOS board name? there are multiple Thinkpad 13's

that0nepers0n, 2022-05-09:
Sorry was busy it’s “Skylake”

MrChromebox, 2022-09-09:
not a script issue, closing

closed 2022-09-09 by MrChromebox